sfera-labs/iono #6: Iono Uno modbus app crashes when enabling DI6 (Onewire)

rpinola, 2022-08-28:
When connecting DS18B20 temperature sensor to DI6, setting AO1 to 5000mV  and then reading holding register 6001 to enable it, Iono Uno becomes unresponsive. This is repeatable and happens every time.

How to troubleshoot this?

giampiero7, 2022-08-29:
Hello @rpinola ,
Thanks for your report.
Just a few clarifications:
- Enabling of the DI6 bus is done by register 6000, is it the reading of 6000 or 6001 that causes the hang?
- Does the hang happen even if AO1 is not enabled before?
- What Arduino board are you using?
- Can you confirm you are not reading register 8001 too?
- Does the same happen on DI5?
Thank you!

rpinola, 2022-08-29:
> * Enabling of the DI6 bus is done by register 6000, is it the reading of 6000 or 6001 that causes the hang?

Sorry, I meant register 6000. On the master device I just have to add +1 to every address so thats the reason. So, reading register 6000 causes the hang.

> * Does the hang happen even if AO1 is not enabled before?

No, it only happens when I enable AO1 to 5000mV. So its related to the temp sensor definitely.

> * What Arduino board are you using?

Arduino Uno, which was shipped with the Iono.

> * Can you confirm you are not reading register 8001 too?

Yes, I have never read that register.

> * Does the same happen on DI5?

This I probably have to test later, don't know.

giampiero7, 2022-08-29:
Thanks for your answers!
When you manage to test DI5, could you also make sure that the issue is not caused by only powering the sensor (set AO1 to 5V) without reading the enable register?
Moreover, do you have a different sensor you can test it with?

rpinola, 2022-09-01:
>  When you manage to test DI5, could you also make sure that the issue is not caused by only powering the sensor (set AO1 to 5V) without reading the enable register? Moreover, do you have a different sensor you can test it with?

I have now tested with 2 different sensors, and both with DI6 and DI5. 

The result is always the same: If the sensor is connected to DI5 reading register 5000 will cause a hang. So not sensor or DI5/DI6 depedent.

Reading register 5000/6000 will respond with 1 before it hangs, with sensor connected. So it is related to the sensor being recognized and that causing a hang.

Only powering the sensor does not cause a hang, only register 5000/6000 reading with sensor powered will. This should be easily repeatable, unless I have a bad piece of hardware etc.

giampiero7, 2022-09-02:
Got hold of a 1wire probe and after a few trials I managed to replicate it.
It looks like a memory issue: the latest added features pushed the memory usage on an Arduino Uno to the limit and when memory is allocated to handle the discovered probes things break.

A quick fix is to roll back the the previous version which does not include Wiegand support:
https://github.com/sfera-labs/iono/blob/5e9ce154454cf3049bcdee3ee4b51f23a8dc1dea/IonoModbusRtu/IonoModbusRtuSlave.cpp
just replace your local file with the above.

Let me know if this solves the issue for you too.

On another note, in case you missed it, Iono Uno does not have pull-ups on the bypassed DI5/DI6 lines so, to communicate with the sensor properly, you might need to add an external pull-up resistor between AO1 and DI5-DI6.

rpinola, 2022-09-03:
> Got hold of a 1wire probe and after a few trials I managed to replicate it. It looks like a memory issue: the latest added features pushed the memory usage on an Arduino Uno to the limit and when [...]

Yes this rolling back worked for me, now able to enable and read onewire-sensors. Great! 

Don't really need Wiegand support so its okay for me to use the older version.

Thanks