sfera-labs/iono #8: best way to flash ModbusRtuApp with default configuration

andhya, 2022-10-05:
Is there an intended way to provide a default configuration (baudrate,address,parity,IOrules) in the ModbusRtuApp.ino sketch?

I thought there might be somewhere in the sketch I could just initialise these variables with a default value, but it looks like the way these are stored in EEPROM means this isn't quite so simple?

I'm working for a sustainability company that have asked me to evaluate the IonoMKR units as a potential cost-effective way of attaching calibrated thermometers to their existing modbus networks, so I'm hoping the flashing and configuring can be done inhouse as a single step, rather than manually setting up each MKR Zero board.

Thanks for your help

andhya, 2022-10-05:
I haven't gone through the details of how the states are captured in the bytes, but in principle would it be feasible to abstract the default configuration in getEEPROMConfig() ?

something like;

```
#ifdef IONO_RP
  EEPROM.begin(256);
#endif
  if (!readEepromConfig(&speedCurrent, &parityCurrent, &addressCurrent, rulesCurrent)) {
    speedCurrent = speedDefault;
    parityCurrent = parityDefault;
    addressCurrent = addressDefault;
    for (int i = 0; i < IORULES_LEN; i++) {
      rulesCurrent[i] = rulesDefault;
    }
  }
  return (speedCurrent != 0 && parityCurrent != 0 && addressCurrent != 0);
}
```

and they could be declared right at the top of the sketch as 0/- as shipped, with some comments on how to set them to valid numbers?

giampiero7, 2022-10-05:
Hi @andhya ,
Yes, you can do that, or if you want to disable completely the reading from EEPROM, in `setup()` you could just not call `getEEPROMConfig()`, set your default `speedCurrent` ... `rulesCurrent`, and set `validConfiguration = true;`

closed 2022-10-05 by giampiero7